ad-si/Brickify #568: Add a "Order Lego parts" link to instructions / sell parts

AtiX, 2015-06-08:
individual plates can be ordered here:
http://shop.lego.com/de-DE/Pick-A-Brick-NachThemenwelt
or here: http://www.brick-shop.de/index.php/cat/c378_Platte-1x6-Platte-1x6-Einzelsteine-online-3666-shop-1-reihig-Ersatzteile-LEGO-Pickabrick.html

cool would be a single button order, where the list would be transfered to the shop and a pre-filled shopping cart would be generated. maybe we can get some sort of a reflink. maybe we can offer a order system and write a script to order them for us and make ~10% profit or whatever :D....

opinions?

Steditor, 2015-06-09:
I'm not sure about this - it does not exactly match the idea of faster printing if one has to wait for ordered bricks to arrive...

adius, 2015-06-09:
:dollar: :heavy_dollar_sign: :moneybag: :money_with_wings:

adius, 2015-06-09:
There is an LEGO affiliate program http://shop.lego.com/en-US/Affiliate-Program