ad-si/Brickify #644: Layouter is too greedy

AtiX, 2015-06-23:
1. Make everything 3d-print
2. brush the bottom layer of the object (e.g. cube.stl) with the make lego brush
3. often: 3-layered bricks are generated even though not all of these layers were selected for LEGO

AtiX, 2015-06-23:
Only the bottom voxel layer was selected: 

[image]

johannesdeselaers, 2015-06-23:
hmm, i've been able to reproduce this once (out of about 70 tries), i.e. it's pretty hard to debug that way
any particular way you were brushing that causes the issue more often?

stuikomma, 2015-06-24:
Just brush the ground level in one stroke. Worked every time. Make sure to brush only that layer and no voxel in layer 1.

johannesdeselaers, 2015-06-24:
there seems to be a general issue when placing LEGO at a voxel not connected to any other curretly existing LEGO
i am still searching for the root cause

johannesdeselaers, 2015-06-26:
update:
i believe the error only occurs after using the 'make everything 3D-print' functionality (please verify)
the reinitialization of the grid seems broken, still figuring out why

johannesdeselaers, 2015-06-30:
replaced by #682

closed 2015-06-30 by johannesdeselaers